cogu/autosar #25: run ar4 tutorial2\components failed as 'Workspace' object has no attribute 'defaultPackages'

parai, 2018-06-16:
Hi cogu:
 I am parai, and I am using this project branch feature/ar4_api for my own AUTOSAR project https://github.com/parai/as which is a study platform of AUTOSAR with python toolchain support for the BSW configuration and CAN COM/DCM/BOOTLOADER tools.
Now I am trying to use your project as RTE configuration and related C source code generator. So I started from the tutorial2, but I failed as below exception, could you help to have a look and how to fix this issue.

And the major feature I want is the RTE C source code generator, can you provide much more demo on this?

D:\repository\as\release\download\autosar\tutorial2\components>python RealTimeClock.py
Traceback (most recent call last):
  File "RealTimeClock.py", line 35, in <module>
    ws.apply(RealTimeClock)
  File "../..\autosar\workspace.py", line 446, in apply
    template.apply(self)
  File "RealTimeClock.py", line 12, in apply
    package = ws.getComponentTypePackage()
  File "../..\autosar\workspace.py", line 510, in getComponentTypePackage
    packageName = self.defaultPackages["ComponentType"]
AttributeError: 'Workspace' object has no attribute 'defaultPackages'

# manually add one line code after ws created "package=ws.createPackage("ComponentType", role="ComponentType")", but still get below error
D:\repository\as\release\download\autosar\tutorial2\components>python RealTimeClock.py
Traceback (most recent call last):
  File "RealTimeClock.py", line 36, in <module>
    ws.apply(RealTimeClock)
  File "../..\autosar\workspace.py", line 446, in apply
    template.apply(self)
  File "RealTimeClock.py", line 15, in apply
    cls.addPorts(swc)
  File "RealTimeClock.py", line 21, in addPorts
    swc.apply(Signals.RtcHours.Send)
  File "../..\autosar\component.py", line 120, in apply
    template.apply(self)
  File "..\SignalsHelper.py", line 26, in apply
    _createProvidePortHelper(swc, cls.name, cls.portInterfaceTemplate, cls.initValueTemplate)
  File "..\SignalsHelper.py", line 7, in _createProvidePortHelper
    ws.apply(portInterfaceTemplate)
  File "../..\autosar\workspace.py", line 446, in apply
    template.apply(self)
  File "..\PortInterfaces.py", line 10, in apply
    package = ws.getPortInterfacePackage()
  File "../..\autosar\workspace.py", line 480, in getPortInterfacePackage
    package = self.find(self.defaultPackages["PortInterface"])
AttributeError: 'Workspace' object has no attribute 'defaultPackages'

cogu, 2018-06-25:
Hello,
The tutorial2 directory contains old files from the [autosar-demo project](https://github.com/cogu/autosar-demo). They reference deprecated and partially removed functionality from the Workspace class.

The attribute defaultPackages has been explicitly removed from the workspace and will not be added again to the main branch. The new template class mechanism replaces defaultPackages. 

The RTE generator in its current state is highly experimental and only supports AUTOSAR 3 models. I assume you are trying to use it for some AUTOSAR 4 project? This will simply not work as it does not support the AUTOSAR 4 type system.

parai, 2018-06-27:
HI Cogu:
Thanks for your comments.

In fact I didn't have an open source AUTOSAR 4 bsw, I still reuse arccore V3.x and I am trying to integrate your AUTOSAR 4.x RTE generator, I am also not sure what AUTOSAR 4.x RTE looks like.

According to my opinion, I think RTE is a series of API interface generted and mapped to BSW. I just need some good demos to learn what it is and how to use it, if possible can you provide a demo?

cogu, 2018-06-27:
The RTE layer is primarily there to support the application software components (SWCs), not the BSW components. 
This Python module has zero focus on BSW integration or any interaction with BSW whatsoever. It's primary purpose is to help developers quickly create application SWCs, the XML files that lives above the RTE layer.

The little BSW Python code I have is entirely there to help the SWC developer test his or her SWC in a simulated environment on PC before integrating on target (which still requires a commercial AUTOSAR toolchain). Pretty much all my github repos exist to make this simulation initiative happen sometime in the future. You can read a little bit more about it [here](https://github.com/cogu/amber). There are currently no working demos of this.

cogu, 2019-11-12:
I removed the tutorial2 directory since that code is now obsolete

closed 2019-11-12 by cogu