cogu/autosar #27: the demo always has error,about module 'autosar' has no attribute 'workspace'

Eng2Mot, 2018-07-24:
the demo always has error,about module 'autosar' has no attribute 'workspace'

Traceback (most recent call last):
  File "C:\Users\gz03232\Documents\autosar.py", line 1, in <module>
    import autosar
  File "C:\Users\gz03232\Documents\autosar.py", line 3, in <module>
    ws = autosar.workspace(version='4.2.2')
AttributeError: module 'autosar' has no attribute 'workspace'
>>> 

demo application as follow:
import autosar
    
ws = autosar.workspace(version='4.2.2')
        
ws.createPackage('DataTypes', role='DataType')
ws.createPackage('Constants', role="Constant")
ws.createPackage('PortInterfaces', role="PortInterface")
ws.createPackage('ModeDclrGroups', role="ModeDclrGroup")
ws.createPackage('ComponentTypes', role='ComponentType')
  
#save the file as XML
ws.saveXML('Workspace.arxml')

    
ws = autosar.workspace(version='4.2.2')
        
ws.createPackage('DataTypes', role='DataType')
ws.createPackage('Constants', role="Constant")
ws.createPackage('PortInterfaces', role="PortInterface")
ws.createPackage('ModeDclrGroups', role="ModeDclrGroup")
ws.createPackage('ComponentTypes', role='ComponentType')

#save workspace packages into XML files
ws.saveXML('DataTypes.arxml', filters=['/DataTypes'])
ws.saveXML('Constants.arxml', filters=['/Constants'])
ws.saveXML('PortInterfaces.arxml', filters=['/PortInterfaces', '/ModeDclrGroups'])
ws.saveXML('ComponentTypes.arxml', filters=['/ComponentTypes'])

cogu, 2018-07-26:
It seems you have named your own script file autosar.py and the Python engine seems to use that one as the actual python module "autosar".

Try renaming your script file to anything except "autosar.py", like "demo.py"

cogu, 2019-11-12:
This issue is no longer relevant

closed 2019-11-12 by cogu

Sudhanva10, 2023-11-08:
@cogu Thanks man.
It looks like the solution which you gave is still relevant.