cogu/autosar #28: Error on Version and Schema for autosar 4

DeathThenNothing, 2018-08-03:
(major, minor, patch, schema) = parseAutosarVersionAndSchema(xmlroot)
ValueError: not enough values to unpack (expected 4, got 3)

ar4 arxml:
"<AUTOSAR xmlns="http://autosar.org/schema/r4.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://autosar.org/schema/r4.0 AUTOSAR_00044.xsd">"

DeathThenNothing, 2018-08-03:
It is ok with autosar_4-3-0 instead of AUTOSAR_00044

cogu, 2018-08-03:
Versioning is defined in an AUTOSAR specification named "ARXML Serialization Rules". You are right in that it seems like they changed the definition in AUTOSAR 4.3 to allow such numbers as AUTOSAR_00044.xsd.

As I currently only support up to AUTOSAR 4.2 my Python module will not handle it.

I will keep this issue open for a while until I have time to implement the new versioning scheme.

Thanks for notifying me about the error.

sperly, 2020-02-26:
I also noticed this today. We also use 4.3.1 with the AUTOSAR_00044.xsd

cogu, 2024-03-29:
* All releases prior to v0.5.0 supports old format only (major, minor patch)
* Starting from v0.5.0 only the new format is supported (schema-version as an integer)

closed 2024-03-29 by cogu