cogu/autosar #30: Init value for all data elements

DeathThenNothing, 2018-08-17:
I have something like this:
swc.createRequirePort(initValue = [0,1,1,0])
In the port I have 4 data elements and i want to put init value for each, or i want to put something like this: 
{0,0,0,0}.
Seems that the package take only the first element 0)
                   <INIT-VALUE>
                    <NUMERICAL-VALUE-SPECIFICATION>
                      <SHORT-LABEL>0</SHORT-LABEL>
                      <VALUE>0</VALUE>
                    </NUMERICAL-VALUE-SPECIFICATION>
                  </INIT-VALUE>

cogu, 2018-08-20:
You're probably right. The company that I work at that is using this technology has a policy to use only a single data element per port. Therefore I have not fully implemented support for multiple data elements as it hasn't been requested.

cogu, 2019-11-12:
Multiple data elements is now possible in release v0.3.6. Please check the unit test suite for examples.

closed 2019-11-12 by cogu