cogu/autosar #31: Support for AUTOSAR 4.4.0 ?

renukavinay, 2019-01-04:
Hello @cogu 
I am new to the AUTOSAR world and was wondering do you have any plans to support AUTOSAR 4.4.0 version?  Many thanks in advance.

cogu, 2019-01-05:
Hi @renukavinay,

I'm fully engaged working on my C-based repositories and will not have time to do any major work on the autosar repo in the next 3-6 months.
I do plan to resume working on the autosar repo at some point in the future and will at that time look into supporting AUTOSAR 4.3 and 4.4 XML.

renukavinay, 2019-01-05:
I am new to autosar and found your python package online and wanted to
check for latest release. Thanks for your response, I really appreciate it.

Regards,
Renuka

purnimsinha, 2019-11-21:
hey, I am new to autosar and i saw your python autosar package and i want to use can you tell me how to install this autosar module in python? many thanks in advance.

cogu, 2019-11-21:
This Python module does not yet support AUTOSAR 4.3 or 4.4.
If you want to use it anyway check out the "Getting Started" guide at https://autosar.readthedocs.io/en/latest/start.html

miketsukerman, 2020-10-12:
Hi @cogu, I would like to contribute to your library to bring support for AUTOSAR 4.3 standard, at least.

cogu, 2020-10-15:
@miketsukerman, Thank you for your support.
I think the first step would be is to properly handle the new versioning format of the XML schema.
Before AUTOSAR 4.3.1 you could read out major, minor and patch version from the root XML element.
Example
```
xsi:schemaLocation="http://autosar.org/schema/r4.0 AUTOSAR_4-3-0.xsd"
```

Starting from AUTOSAR 4.3.1 they are only using the revision number of the release in the XML root element
Example:
```
xsi:schemaLocation="http://autosar.org/schema/r4.0 AUTOSAR_00044.xsd"
```

You can read more about this in chapter *2.2.4 XML Root Element* of the AUTOSAR specification AUTOSAR_TPS_ARXMLSerializationRules.pdf

Traditionally, a workspace is created by giving the *version* argument to the workspace method.
```python
ws = autosar.workspace(version="4.2.2")
```

That API should continue to work but In addition we need to be able to create workspaces based on release number instead of version.

```python
ws1 = autosar.workspace(release=44) #AUTOSAR 4.3.1

ws2 = autosar.workspace(release=46) #AUTOSAR 4.4.0
```

What I would like to see first is a set of Python unit tests that can demonstrate that the Python module is capable of serializing an (empty) AUTOSAR 4.3.1 workspace into ARXML and then reads it back again using the parser submodule.

Sort of like this:
```python
def test_write_and_read_empty_autosar_431_file(self):
    ws1 = autosar.workspace(release=44)
    file_name = 'ar431_empty_workspace.arxml'
    generated_file = os.path.join(self.output_dir, file_name)
    expected_file = os.path.join( 'expected_gen', 'base', file_name)
    self.save_and_check(ws, expected_file, generated_file)
    ws2 = autosar.workspace()
    ws2.loadXML(os.path.join(os.path.dirname(__file__), expected_file))
    self.assertEqual(ws2.release, 44)
```

I also think that *release* and *version* should be mutually exclusive in the Workspace object such that:
```python
ws1 = autosar.workspace(version="4.2.2")
assert(ws1.release is None)

ws2 = autosar.workspace(release=44)
assert(ws2.version is None)
```

For the above to work, the current implementation needs to change. I do believe that current implementation sets ws.version to `4.0` when it parses ARXML containing the `AUTOSAR_{release}.xsd` format which I believe was a design mistake.

Please let me know what you think.

BR1py, 2021-04-21:
Hello,
can the developers please report if there is any progress regarding the integration of newer AUTOSAR versions 4.3 or 4.4?

Can you please give an estimation? 

We might support you but if there are no plans we must create an own solution.

cogu, 2021-04-21:
@BR1py, 
Work on this project is mostly done on evenings and weekends. That's why progress is so slow.
I have so many other things I want to implement in this project and in others before I get to this part.
I can at least say that supporting AUTOSAR 4.3 will be prioritized before AUTOSAR 4.4 and that I see it as unlikely this work will start during 2021.

cogu, 2024-03-29:
* Releases up until v0.4.1 supports AUTOSAR 3.0 and 4.2.2 only.
* Releases from v0.5.0 onwards is verified against AUTOSAR schemas in the range 4.5 to 4.8.
* AUTOSAR 4.3 and 4.4 might work for some elements but won't be officially supported.

closed 2024-03-29 by cogu