cogu/autosar #32: use lxml to generate arxml?

grissiom, 2019-04-24:
With a glance of the code, I found you generate the arxml by hand. Is there any plan to port the writer to lxml? I personally use lxml to generate arxml and the `pretty_print` can perfectly fit the AUTOSAR indentation standard.

cogu, 2019-04-24:
If there is a significant speed improvement involved I will certainly look into it.

grissiom, 2019-04-26:
I've not benchmark your project yet, but in my personal usage, it takes just seconds to load/parse/modify/write a ~5MB ARXML.

Besides the speed, I think the code will be much more cleaner if we switch to lxml.builder.ElementMaker API: https://lxml.de/api/lxml.builder.ElementMaker-class.html .

If you are interested, maybe I could do the porting and send you pull request when it's ready.

cogu, 2019-04-26:
That's OK, I will run some experiments soon. I'm currently in the middle of a major update that I will commit in a couple of days. It's important that the ARXML is recreated binary same as the Vector toolchain it's replacing. This is the reason why the autosar.writer package is handcrafted today.

As for current performance, my Python lib can both load and write ARXML at about 170k lines/s or about 14MB worth of ARXML/s (On my core i7 laptop). This is about 10x faster than the current Vector toolchain.

closed 2019-04-26 by cogu