cogu/autosar #38: Description attribute in record elements

Iskander1986, 2019-10-25:
I found a mismatch behavior, for tooling convention like a System Desk from DSpace it's possible to create a description attribute inside Record elements.
So to solve those issue i suggest to add between line 84 and 85 the following code:

```
tmp = self.writeDescXML(childElem)
if tmp is not None: lines.extend(self.indent(tmp,3))
```
This is the same code are used in other classes.

cogu, 2019-11-25:
This is currently not prioritized and will be fixed later. If you need the fix faster you are welcome to create a pull-request.

cogu, 2024-03-29:
* This won't be fixed in version 0.4.x.
* In version 0.5.x the desc element is supported equally well for all elements due to its new class design.

closed 2024-03-29 by cogu