cogu/autosar #41: NV_Data_Interface

Tofeni, 2019-12-10:
Hello,

i have to generate an arxml file, which will be imported in a davinci workspace.

I can not find a function to create an NV_DATA_INTERFACE element.
Can someone tell me if this can be done or not, and if it will be implemented in a future version?

Here is a model how it is look like:
[image]

Thank you,
Marius

cogu, 2019-12-11:
Hi Marius,
I'm mostly using the AUTOSAR compendium for reference (https://www.ar-compendium.com)
What you are talking about is chapter 7.7 "Nonvolatile Data Port"
This means you trying to create a special kind of software component called "NVBlock Software Component".
This Python module does not support anything from chapter 7.7 yet and I have no current plants to implement it. Progress on this module is slow since I'm doing the work in my spare time.

Anyone is welcome to fork this repo and implement it on your own using pull request.
However, I will only accept pull requests that have working unit tests cases for the new feature.

Tofeni, 2020-01-06:
@kallemooo , Thank you for your pull request!

I have one more question:

How can i crate a nvdata with type: "VARIABLE-DATA-PROTOTYPE"?

cogu, 2020-01-07:
Check the file ar4_portinterface_test.py for reference.
There should be two test methods that you can use as examples: test_create_nv_data_interface_single_element, test_create_nv_data_interface_multiple_elements

Tofeni, 2020-01-07:
@cogu Thank you!

cogu, 2020-01-07:
@kallemooo 
Thanks for adding the test cases

closed 2020-01-07 by cogu