cogu/autosar #54: <PR-PORT-PROTOTYPE>

Tofeni, 2020-01-17:
Hello,

i have 3 questions:

1) Is there a way to add <DESC> to an element?
2) It is possible to create Provider receiver port?
3) Why i can not crate ProvidePort if i have many elements in a PortInterface?

[image]

Thank you,
Tofeni

cogu, 2020-01-17:
Question 1: I'm not sure what you mean. Is it related to question 3?
Question 2: No, PR-PORT-PROTOTYPE has not been implemented in this Python package.
Question 3: You can certainly have many data elements in an interface. See test case test_create_nvblock_software_component in file ar4_component_test.py for example. 
@kallemooo Thanks for providing that test case.

Tofeni, 2020-01-17:
1) at the first question i write <  desc> (DESC)

3) and at the 3th, i am trying this:
swc.createProvidePort('P_DiagNv_ParameterHandler_Cfg_PRDiagNv_ParameterHandler', '/PortInterfaces/IfDiagNv_ParameterHandler_Cfg', romBlockInitValue='0'), and i get next error:
[image]

i will try with the test case test_create_nvblock_software_component in file ar4_component_test.py

Thank you

cogu, 2020-01-21:
Description fields are not implemented for ports.

Some time ago I refactored the XML parser to share common parsing code for tags that occurs often.

I will need to refactor the XML writer in a similar fashion so that it automatically handles common fields such as desc and admin-data without having to add them manually for every single XML element.

cogu, 2024-03-29:
* Starting from v0.5.3 PR-PORTS are now supported
* Description elements are also supported for all elements, including ports.

closed 2024-03-29 by cogu