cogu/autosar #67: The example runs with error:    d = Decimal(float(version)) ValueError: could not convert string to float: '4.2.2'

DDxxg, 2020-04-14:
Hi, I use the example like as bellows:
import autosar

#Create some packages and data types

ws = autosar.workspace('4.2.2')
datatypes = ws.createPackage('DataTypes', role='DataType')
basetypes = datatypes.createSubPackage('BaseTypes')
u8Type = basetypes.createSwBaseType('uint8', 8, nativeDeclaration='uint8')
u16Type = basetypes.createSwBaseType('uint16', 16, nativeDeclaration='uint16')
u32Type = basetypes.createSwBaseType('uint32', 32, nativeDeclaration='uint32')
ws.saveXML('basetypes.arxml')

#Print reference and name for the created objects
print('u8Type.name: ' + u8Type.name)
print('u8Type.ref: ' + u8Type.ref)
print('u16Type.name: ' + u16Type.name)
print('u16Type.ref: ' + u16Type.ref)
print('u32Type.name: ' + u32Type.name)
print('u32Type.ref: ' + u32Type.ref)

and it fails with:
    d = Decimal(float(version))
ValueError: could not convert string to float: '4.2.2'

cogu, 2020-04-14:
Hi, I am unable to reproduce this error. I have tried both master branch (latest) as well as release v0.3.8 and they both work fine with the given example.

1. Which release/commit are you actually using?
2. What is your Python version and Operating System?
3. Is your OS using a locale/language where floating point separator isn't the dot character (.)?

Also, could you try this example and report back what it prints?
---
import autosar

ws = autosar.workspace('4.2.2')
print(ws.version)
---

It's supposed to print "4.2" but in your case it seems to have become "4.2.2" which means that version parser isn't working properly to begin with. Hence question 3 above.

cogu, 2024-03-29:
Unable to reproduce. Closing due to inactivity.

closed 2024-03-29 by cogu