cogu/autosar #68: AUTOSAR NVM configuration

Jihao-Hu, 2020-04-22:
Now to implement such a function when writing data to the NVM after shutdown, a wake-up source is detected, and the controller will wake up again at this time the ECU will be reseted again.  and the data in NVM is incomplete. At this time, the RAM will reload the previous data in NVM after reset, but at this time we do not want to use Nvm_ReadAll () to read the previously stored data from NVM, but want to keep the last unsaved data in RAM

cogu, 2020-04-23:
I only know the very basics of NVM and your question sounds quite complex.
Perhaps @kallemooo who implemented most of the NVM features might know more.
However, my best recommendation is that you direct your question to your AUTOSAR platform vendor directly.

kallemooo, 2020-06-26:
To do this you need to manually handle the read of the NVM data for the specific NVM block.
Fot the NVM block set "RESTORE-AT-START" parameter to false, and use NvM_Read() to fetch data the data from the storage when needed.

But as @cogu writes the recommendation is that you direct your question to your AUTOSAR platform vendor directly.

cogu, 2024-03-29:
Issue out of scope for this project.

closed 2024-03-29 by cogu