cogu/autosar #72: Trouble creating constant

sorinsaladea, 2020-06-30:
Hello,

Is there a way to create the following structure?

    <AR-PACKAGE>
      <SHORT-NAME>Constants</SHORT-NAME>
      <ELEMENTS>
        <CONSTANT-SPECIFICATION UUID="9B2B418A-CA6C-4E3B-9562-D49D1C2695B1">
          <SHORT-NAME>CDiagNv_NvMServiceRequestType</SHORT-NAME>
          <VALUE-SPEC>
            <ARRAY-VALUE-SPECIFICATION>
              <ELEMENTS>
                <RECORD-VALUE-SPECIFICATION>
                  <FIELDS>
                    <NUMERICAL-VALUE-SPECIFICATION>
                      <SHORT-LABEL>ServiceId</SHORT-LABEL>
                      <VALUE>0</VALUE>
                    </NUMERICAL-VALUE-SPECIFICATION>
                    <NUMERICAL-VALUE-SPECIFICATION>
                      <SHORT-LABEL>RequestResult</SHORT-LABEL>
                      <VALUE>0</VALUE>
                    </NUMERICAL-VALUE-SPECIFICATION>
                  </FIELDS>
                </RECORD-VALUE-SPECIFICATION>

I tried creating array data type, record data type and RECORD-VALUE-SPECIFICATION and ARRAY-VALUE-SPECIFICATION are reversed.

cogu, 2020-08-10:
Hello @sorinsaladea,
I just tried this myself and it seems array of records is not yet implemented. I will create fix soon.

closed 2020-08-23 by cogu

cogu, 2020-08-23:
@sorinsaladea,

Here is a complete example to re-create that XML:
```python
ws = autosar.workspace(version="4.2.2")
package = ws.createPackage('Constants', role='Constant')
package.createConstant('CDiagNv_NvMServiceRequestType', None,
    [
           {'SerciceId': 0, 'RequestResult': 0}
    ], label = None)
ws.saveXML('Constants.arxml', filters=['/Constants'])
```

PR #73 (2020-08-23): Possible to create constants without data type reference
Possible to create array of record values.

Fixes #72