cogu/autosar #74: Support for nested subElement in ImplementationDataType and ImplementationDataTypeElement

insubnia, 2020-09-28:
Hi @cogu 

Before starting, I'm big fan of your autosar module and it's very useful for my works.
As far as i know, ImplementationDataTypeElement can have subElement like belows.

           <IMPLEMENTATION-DATA-TYPE UUID="ffc534fa-9e89-4a14-82a1-eff8e4acaf50">
              <SHORT-NAME>Dt_RECORD_Reaction</SHORT-NAME>
              <CATEGORY>STRUCTURE</CATEGORY>
              <SUB-ELEMENTS>
                <IMPLEMENTATION-DATA-TYPE-ELEMENT UUID="931668fd-65e0-46f9-8768-fc6760c494c1">
                  <SHORT-NAME>FCA</SHORT-NAME>
                  <CATEGORY>STRUCTURE</CATEGORY>
                  <SUB-ELEMENTS>
                    <IMPLEMENTATION-DATA-TYPE-ELEMENT UUID="b56869c7-2039-4421-ad04-994f7773eebd">
                      <SHORT-NAME>Ro_Inhibit</SHORT-NAME>
                      <CATEGORY>TYPE_REFERENCE</CATEGORY>
                      <SW-DATA-DEF-PROPS>
                        <SW-DATA-DEF-PROPS-VARIANTS>
                          <SW-DATA-DEF-PROPS-CONDITIONAL>
                            <IMPLEMENTATION-DATA-TYPE-REF DEST="IMPLEMENTATION-DATA-TYPE">/AUTOSAR_DataTypes/ImplementationDataTypes/uint8</IMPLEMENTATION-DATA-TYPE-REF>
                          </SW-DATA-DEF-PROPS-CONDITIONAL>
                        </SW-DATA-DEF-PROPS-VARIANTS>
                      </SW-DATA-DEF-PROPS>
                    </IMPLEMENTATION-DATA-TYPE-ELEMENT>
                    <IMPLEMENTATION-DATA-TYPE-ELEMENT UUID="20ed6f15-f125-4d10-88db-7346cf01e403">
                      <SHORT-NAME>Sf2_Inhibit</SHORT-NAME>
                      <CATEGORY>TYPE_REFERENCE</CATEGORY>
                      <SW-DATA-DEF-PROPS>
                        <SW-DATA-DEF-PROPS-VARIANTS>
                          <SW-DATA-DEF-PROPS-CONDITIONAL>
                            <IMPLEMENTATION-DATA-TYPE-REF DEST="IMPLEMENTATION-DATA-TYPE">/AUTOSAR_DataTypes/ImplementationDataTypes/uint8</IMPLEMENTATION-DATA-TYPE-REF>
                          </SW-DATA-DEF-PROPS-CONDITIONAL>
                        </SW-DATA-DEF-PROPS-VARIANTS>
                      </SW-DATA-DEF-PROPS>
                    </IMPLEMENTATION-DATA-TYPE-ELEMENT>
                  </SUB-ELEMENTS>
                </IMPLEMENTATION-DATA-TYPE-ELEMENT>
              </SUB-ELEMENTS>
            </IMPLEMENTATION-DATA-TYPE>

Is it possible to generate above code by using your module? If not, you have some plans for support? (I hope module had createImplementationDataElement APIs)
Thank you in advance

cogu, 2020-10-04:
Do you mean STRUCTURE element inside a STRUCTURE element? Yeah, that's currently not supported. You're right that the datatype.ImplementationDataType class would benefit from a createImplementationDataElement method to make it more powerful.
I'm currently engaged in other projects and don't know when I will have time to implement this feature. I will keep it mind once I get back to working with the Python module.

cogu, 2024-03-29:
Starting from [v0.5.0](https://github.com/cogu/autosar/releases/tag/v0.5.0) it should now be possible now to create record and array data types of arbitrary complexity and depth.

closed 2024-03-29 by cogu