cogu/autosar #79: DEXT format

sorinsaladea, 2021-04-27:
I would like to generate a DEXT (Autosar Diagnostic Extract) like file using Python and one library. 
Is there an extension to this library or something similar that can help me? 

Thank you!

cogu, 2021-04-30:
Hello @sorinsaladea. I'm not aware of any other Python projects that can help you. I did not know there even was a format called DEXT. I used to work with the ODX format a long time ago before I started this AUTOSAR Python project.

ubmarco, 2022-02-09:
DEXT is the diagnostic extract and is part of the system templates, see AUTOSAR_TPS_DiagnosticExtractTemplate.pdf in the 'Methodology and Templates' part of the Autosar Classic docs. It also uses ARXMLs and follows the XSD schema of the Autosar version used. More information can be found in AUTOSAR_TR_Methodology.pdf when searching for 'Diagnostic Extract'.

When looking at https://github.com/cogu/autosar/tree/master/autosar I feel there will be a lot of classes missing that are required by DEXT. The root node of the DEXT is the `DiagnosticContributionSet` which is already missing. I feel this library mainly targets SWCs.

cogu, 2024-03-29:
DEXT format won't be supported. This project is primarily for modelling SWCs.

closed 2024-03-29 by cogu