cogu/autosar #88: multiply arxmls in one workspace

Han40, 2022-04-03:
i want to load create RPort in A workspace's componentType, using Port interface in another arxml.
how to create one workspace with multiply arxmls.
how to load multiply arxmls into one workspace.

 thanks

gcbb, 2022-04-03:
先用wks1加载arxml1
再用wks2加载arxml2
在wks2中直接引用wks1的portinterface就行

Han40, 2022-04-03:
> 先用wks1加载arxml1 再用wks2加载arxml2 在wks2中直接引用wks1的portinterface就行

目前看下来没法 做到 跨 workspace 的方法来 引用 portInterface

gcbb, 2022-04-03:
我百分百肯定可以的
查找出wks1的port interface路径，然后在wks2中引用
我就是这么干的

Han40, 2022-04-04:
如何引用呢，直接路径名路径名就可以吗？ 还是要备注上 wks1:PortInterfaceRef ?
我看了代码里似乎一直在找 自己workspace 内的 PortInterfaceRef
可以看下源码里compoent.py 里的 createRequirePort 方法
我不知道你这边如何实现的呢？？？

gcbb, 2022-04-09:
报歉，我查了下我自己的，，，我是在一个wks中，，，

我只是加我了两个arxml

cogu, 2024-03-29:
Already supported. Not sure what the issue really was.

closed 2024-03-29 by cogu