cogu/autosar #93: Elements not imported when Loading package

palpicture, 2023-04-05:
Hi, First thanks to the contributors for this lib. it's very useful.
I have an issue when loading ARXML file. when i use loadpackage the package is correctly loaded but the elements are staying empty. In the same time in the subpackages elements are loaded perfectly.
If somebody know what is my mistake i will be thankfull ! 
here the simple code that I a use for the loading : 
`ws = autosar.workspace("4.1.1")
ws.openXML("myfile.arxml")
for role in ws.listPackages():
     ws.loadPackage(role)`

I tried with differents files but I have the same issue each time

cogu, 2023-04-05:
Have you tried to just load the XML?
```
ws = autosar.workspace()
ws.loadXML("myfile.arxml")
```

palpicture, 2023-04-07:
Yes but sadly the result is the same

gcbb, 2023-04-07:
> Yes but sadly the result is the same

could you give your arxml ,I will try to

palpicture, 2023-04-07:
here it is 

[myfile.arxml.txt](https://github.com/cogu/autosar/files/11179357/myfile.arxml.txt)

(I add a .txt to be able to upload it on github)

cogu, 2023-04-08:
@palpicture, your file isn't AUTOSAR 4.1.1, it's 4.4.1. On the surface it looks like it's Adaptive AUTOSAR as it has several element types this Python module doesn't support.
Examples of unsupported elements:
* STD-CPP-IMPLEMENTATION-DATA-TYPE (I didn't even know such an element existed)
* SERVICE-INTERFACE (This is for sure Adaptive AUTOSAR)

Other elements such as `APPLICATION-PRIMITIVE-DATA-TYPE` and `COMPU-METHOD` should be parsed, can't see why not. But even if they were parsed I doubt it would be of much use.

closed 2023-04-12 by cogu