cogu/autosar #102: 0.5.2: Type hint for make_packages should be "*refs: str"

qg020, 2024-02-14:
Hi all, great package - thank you!  

I am updating a script that was at 0.4 with PyCharm.  The make_packages call takes a comma separated list of package name strings but the type hint in the function definition says '*refs: list[str]'.  This causes PyCharm to flag the call with a warning since I am not passing a list of strings but the comma separated strings themselves.

Let me know if you'd like me to make a pull request with this change.  

Thank you.

cogu, 2024-02-20:
Thanks for finding the issue, I will fix it soon.

cogu, 2024-03-29:
Fixed in [v0.5.3](https://github.com/cogu/autosar/releases/tag/v0.5.3).
Actually, I now recommend that you use the new method `Workspace.create_package_map` instead. It internally calls make_packages.
I've updated all the example code and documentation to use the new method.

closed 2024-03-29 by cogu