cogu/autosar #103: 0.5.2 Feature Request: Reader parse-only mode.

qg020, 2024-02-14:
I am reading ARXML we have generated and it is not to the Reader's liking.  I have iterated through fixing each raised exception.  A nice feature would be a 'parse-only' flag that would return a list of all lines in an ARXML file needing attention and why it is a problem.

cogu, 2024-02-20:
Does the parsing stop because an exception is raised by the parser? In that case I can add an option to continue parsing and not stop by raising an error.

Also, have you verified your ARXML file against the schema file before trying to parse?
I always use xmllint tool to verify any XML file first. I always check against schema versions 48, 49, 50 and 51. If your ARXML uses any other version I cannot give any support.

Finally, the new code base is still in early development. I haven't tried it myself yet to parse any XML generated by other tool or script. I will start testing it later, somewhere after v0.5.4.

qg020, 2024-02-20:
Hi cogu,  thanks for the response.  ARXML input files mention r4.0 AUTOSAR_4-2-2.  The problem is definitely in our ARMXL which have duplicate entries.  All errors reported by the Reader indicate that.  Which is great.  Just a convenience thing if you could ask to see all the parsing problems at once.  

Low hanging fruit for when you have time (haha).   Thanks!

cogu, 2024-03-29:
When I have time I plan to add a boolean flag to the reader class named something like "stop_on_error" which defaults to True.
Manually setting it to False should get the effect you want.

But before I can add that I have to refactor all the places in the Reader code that raises an exception to only print a warning based on this flag. It's gonna take some time to do.

rcd09, 2024-03-30:
Hi Conny,

Thank you for giving this some thought and committing to add it.  If you
don't mind, I'll take a crack at making the changes you propose and see if
I can make progress.

-r

cogu, 2024-03-31:
@rcd09, Actually when I think about it I would prefer to keep most of the exceptions and instead catch them in a try-except block in the method `_read_package_elements`.

* If the element is too broken to parse then we just skip it and move on to the next element
* If the element is a duplicate, as seems to be the case for @qg020 then the line `package.append(element)` inside `_read_package_elements` will raise an exception instead, which we can catch in the same place.

I can make the change to the `_read_package_elements`. It should be a quick fix.

cogu, 2024-03-31:
@qg020, @rcd09, I've made a first attempt of the change on latest master. There is still a lot of improvements that can be made but it should be good enough for now.

I decided to let stop_on_error be False as default. Now you have to explicitly set it to True to stop reading the XML.

See [print_errors.py](https://github.com/cogu/autosar/blob/master/examples/xml/reader/print_errors.py) example with an invalid ARXML containing duplicates.

cogu, 2024-10-28:
Resolved in [v0.5.4](https://github.com/cogu/autosar/releases/tag/v0.5.4l)
By default, XML resumes parsing on the next element if an error occurs.

closed 2024-10-28 by cogu