cogu/autosar #104: SWC-IMPLEMENTATION / CONSTANT-SPECIFICATION-MAPPING-SET / APPLICATION-SW-COMPONENT-TYPE support

WeWonderMax, 2024-03-06:
Hi ! 

First of all thank you for your work on this project, helped me quite a lot.

I'm trying to build a "ARXML checker" that would basically **compare 2 ARXML** that should be identical.

I based my work on your object-oriented logic by simply **iterating over each ports** (previously extracted in a dict since your package does not support that feature yet as i understand) and **getting each sub-elements with nested loops**.

So far I processed many (sub) elements and it is working as expected. 

Nonetheless I have a few questions and I hope you can help me out:

- Based on the title, do you plan tu spport these elements in future release? **Could you assess on the time it would take ?** 
- I'm not a python expert, so I wonder how do you feel about my implementation, at first glance **do you think there is a better way to handle things**? 

Thank you for your time, have a good one.

cogu, 2024-03-07:
Which release are you using? v0.41? v0.5.2?
For v0.4.1 SwcImplementation and ApplicationSwComponent are already implemented.
For v0.5.2 they are not, but also ports are not implemented so I have no idea what you mean by the looping thing. Are you looping over elements?

ApplicationSwComponent will be (partially) supported in v0.5.3, SwcImplementation will be supported in v0.5.4.
I have never seen the constant-specification-mapping-set element before. Not sure when I will implement that but it will be sometime after v0.5.4 and before v0.6.0.

WeWonderMax, 2024-03-12:
i use v0.5.2 
basically i extract the ports like this first with xpath and all:

{
            "Name": "svcRearWiperControlMain_setRearWiperIOCtrl",
            "Interface": "/ARRoot/org/rnm/PortInterfaces/If_RearWiperControlSvcProv_setRearWiperIOCtrl",
            "Init_value": "/ARRoot/org/rnm/ConstAppl/initValueDef_If_RearWiperControlSvcProv_setRearWiperIOCtrl",
            "Data_element": "/ARRoot/org/rnm/PortInterfaces/If_RearWiperControlSvcProv_setRearWiperIOCtrl/data"
        },

then your module comes into play to find the refs (document.find) and sub elements (operations,elements....)

I'm looking forward to the next versions as it would be a complete solution for me

cogu, 2024-03-29:
* ApplicationSoftwareComponentType and CompositionSwComponentType is supported in v0.5.3
* SwcInternalBehavior is planned to be incrementally supported in v0.5.4 and v0.5.5
* The ConstantSpecificationMapping element might get support in v0.5.6 (not really planned yet).

closed 2024-03-29 by WeWonderMax