cogu/autosar #109: MultiLanguageReferrable.__init__() got an unexpected keyword argument 'admin_data'

WeWonderMax, 2024-04-18:
`<AR-PACKAGE>
                  <SHORT-NAME>SwComponentTypes</SHORT-NAME>
                  <ELEMENTS>
                    <APPLICATION-SW-COMPONENT-TYPE>
                      <SHORT-NAME>swcX</SHORT-NAME>
                      **<ADMIN-DATA>
                        <SDGS>
                          <SDG GID="arxmlgen">
                            <SD GID="property_x">x</SD>
                          </SDG>
                        </SDGS>
                      </ADMIN-DATA>**
                      <PORTS>`

Hello

This element make the reader crash and return None. I just delete it:

`<AR-PACKAGE>
                  <SHORT-NAME>SwComponentTypes</SHORT-NAME>
                  <ELEMENTS>
                    <APPLICATION-SW-COMPONENT-TYPE>
                      <SHORT-NAME>swcPositionLightsMgr</SHORT-NAME>
                      <PORTS>`

and everything is fine but it is a hassle for me. Can you make it so the reader ignores it ? 

Thanks

cogu, 2024-04-19:
@WeWonderMax, thanks for reminding me. I also encountered the issue a while back. 

I made a fix but I forgot to add it to git. I just pushed it to latest master. 

It simply prevents the crash from happening, it doesn't mean ADMIN-DATA is supported.

cogu, 2024-04-19:
Sorry, I can't do that. However, I did increment the module version to v0.5.4a2 in this git commit and pip install knows how to handle these intermediate version numbers.

WeWonderMax, 2024-04-19:
nvm i will dig into that thanks for support again

closed 2024-04-19 by WeWonderMax