cogu/autosar #111: what does NotImplementedError: CATEGORY mean?

mixaobo, 2024-05-16:
Hello team, 

I try to ws.loadXML and got this issue.  Do you have any hints about this one?

``Python312\Lib\site-packages\autosar\parser\system_parser.py", line 38, in parseSystem
    raise NotImplementedError(xmlElem.tag)
NotImplementedError: CATEGORY

cogu, 2024-05-19:
Hello @mixaobo,

That means the <CATEGORY> XML tag was not expected to be found as a sub-element to the <SYSTEM>tag.

The system parser is the least developed feature of the AUTOSAR Python module v0.4.x and an updated version based on the v0.5 redesign isn't yet planned.

You could try to either remove the <CATEGORY> element directly under <SYSTEM> tag from your XML or you could temporarily change line 38 to just say "pass" instead. 

Both options allow the parsing to resume but since that parser is so underdeveloped there's a risk you will get stuck on another error triggered just after it.

gcbb, 2024-09-11:
我在原来的文件中重写了 CATEGORY的读取， 解析 和回写， 也没啥问题

cogu, 2024-10-28:
Won't be fixed in v0.4.

closed 2024-10-28 by cogu