cogu/autosar #123: tags cannot parse in cp arxml

hogger0, 2024-10-30:
RootComposition3.arxml(16): Unprocessed element <DATA-TRANSFORMATION-SET>
RootComposition3.arxml(52): Unprocessed element <GENERAL-PURPOSE-PDU>
RootComposition3.arxml(64): Unprocessed element <I-SIGNAL-I-PDU>
RootComposition3.arxml(149): Unprocessed element <I-SIGNAL>
RootComposition3.arxml(347): Unprocessed element <SYSTEM-SIGNAL>
RootComposition3.arxml(757): Unprocessed element <SO-AD-ROUTING-GROUP>
RootComposition3.arxml(956): Unprocessed element <SYSTEM>
RootComposition3.arxml(1191): Unprocessed element <ETHERNET-CLUSTER>
RootComposition3.arxml(1802): Unprocessed element <COUPLING-ELEMENT>
RootComposition3.arxml(1827): Unprocessed element <ECU-INSTANCE>

cogu, 2024-10-30:
I've checked the implementation and it's supposed to print the tag name wrapped in "<>" right after "Unprocessed element".

Anyway, it doesn't matter. The message simply means that an XML element is not yet implemented and I am fully aware which elements are not yet implemented in v0.5.

If you don't add additional information to this issue I will close it.

hogger0, 2024-10-30:
`
RootComposition3.arxml(16): Unprocessed element <DATA-TRANSFORMATION-SET>
RootComposition3.arxml(52): Unprocessed element <GENERAL-PURPOSE-PDU>
RootComposition3.arxml(64): Unprocessed element <I-SIGNAL-I-PDU>
RootComposition3.arxml(149): Unprocessed element <I-SIGNAL>
RootComposition3.arxml(347): Unprocessed element <SYSTEM-SIGNAL>
RootComposition3.arxml(757): Unprocessed element <SO-AD-ROUTING-GROUP>
RootComposition3.arxml(956): Unprocessed element <SYSTEM>
RootComposition3.arxml(1191): Unprocessed element <ETHERNET-CLUSTER>
RootComposition3.arxml(1802): Unprocessed element <COUPLING-ELEMENT>
RootComposition3.arxml(1827): Unprocessed element <ECU-INSTANCE>
`

hogger0, 2024-10-30:
RootComposition3.arxml(16): Unprocessed element <DATA-TRANSFORMATION-SET
RootComposition3.arxml(52): Unprocessed element <GENERAL-PURPOSE-PDU
RootComposition3.arxml(64): Unprocessed element <I-SIGNAL-I-PDU
RootComposition3.arxml(149): Unprocessed element <I-SIGNAL
RootComposition3.arxml(347): Unprocessed element <SYSTEM-SIGNAL
RootComposition3.arxml(757): Unprocessed element <SO-AD-ROUTING-GROUP
RootComposition3.arxml(956): Unprocessed element <SYSTEM
RootComposition3.arxml(1191): Unprocessed element <ETHERNET-CLUSTER
RootComposition3.arxml(1802): Unprocessed element <COUPLING-ELEMENT
RootComposition3.arxml(1827): Unprocessed element <ECU-INSTANCE

hogger0, 2024-10-30:
> I've checked the implementation and it's supposed to print the tag name wrapped in "<>" right after "Unprocessed element".
>
> Anyway, it doesn't matter. The message simply means that an XML element is [...]

RootComposition3.arxml(16): Unprocessed element <DATA-TRANSFORMATION-SET
RootComposition3.arxml(52): Unprocessed element <GENERAL-PURPOSE-PDU
RootComposition3.arxml(64): Unprocessed element <I-SIGNAL-I-PDU
RootComposition3.arxml(149): Unprocessed element <I-SIGNAL
RootComposition3.arxml(347): Unprocessed element <SYSTEM-SIGNAL
RootComposition3.arxml(757): Unprocessed element <SO-AD-ROUTING-GROUP
RootComposition3.arxml(956): Unprocessed element <SYSTEM
RootComposition3.arxml(1191): Unprocessed element <ETHERNET-CLUSTER
RootComposition3.arxml(1802): Unprocessed element <COUPLING-ELEMENT
RootComposition3.arxml(1827): Unprocessed element <ECU-INSTANCE

hogger0, 2024-10-30:
These labels are not implemented

hogger0, 2024-10-30:
These labels are not implemented？

cogu, 2024-10-30:
That's right. None of them are implemented yet and won't be for a very long time. Some of the system-related ones isn't planned until v0.8.
You can always check the CHANGELOG to see which elements are currently supported.