cogu/autosar #131: Some tags are missing

ChandlerLiu3, 2025-03-06:
- TRIGGER-INTERFACE, this is the 6th interface type
- SERVICE-SW-COMPONENT-TYPE
- ECU-ABSTRACTION-SW-COMPONENT-TYPE: this one can be read from arxml as an application SWC

```
            'APPLICATION-SW-COMPONENT-TYPE': self._read_application_sw_component_type,
            'ECU-ABSTRACTION-SW-COMPONENT-TYPE': self._read_application_sw_component_type,
```

ChandlerLiu3, 2025-03-06:
Hi, @cogu : can I join and contribute?

cogu, 2025-03-12:
Hi @ChandlerLiu3, you don't seem to have any public Python repositories of your own and I'm therefore unable to assess your Python skill level. Open source developers these days don't seem to bother to even write unit tests.

I'm aware that some XML tags are not implemented. The various ComponentTypes are pretty similar to each other as you have observed.

I just added support for adding both external and internal trigger-points in my last commit (on a branch) but the port interface itself is still not implemented. Will add it soon.

cogu, 2026-08-17:
All three elements and a bunch more are now supported in v0.5.6.

closed 2026-08-17 by cogu